theophile/SuperSlicer_to_Orca_scripts #1: running QA attempts

MAVProxyUser, 2023-08-03:
```
$ perl superslicer_to_orca-print-process.pl --input 
Klipper - Creality Ender-3 S1 Pro (0.4 mm nozzle).ini  Klipper - Creality Ender-5 S1 (0.4 mm nozzle).ini      superslicer_to_orca-print-process.pl
Kevins-MacBook-Air:printer kfinisterre$ perl superslicer_to_orca-print-process.pl --input Klipper\ -\ Creality\ Ender-3\ S1\ Pro\ \(0.4\ mm\ nozzle\).ini --outdir ./ --nozzle-size 0.4
Use of uninitialized value in division (/) at superslicer_to_orca-print-process.pl line 148.
Use of uninitialized value $component in multiplication (*) at superslicer_to_orca-print-process.pl line 148.
Use of uninitialized value in division (/) at superslicer_to_orca-print-process.pl line 148.
Use of uninitialized value $component in multiplication (*) at superslicer_to_orca-print-process.pl line 148.
Use of uninitialized value in division (/) at superslicer_to_orca-print-process.pl line 148.
Use of uninitialized value in division (/) at superslicer_to_orca-print-process.pl line 148.
Use of uninitialized value $component in multiplication (*) at superslicer_to_orca-print-process.pl line 148.
Use of uninitialized value in division (/) at superslicer_to_orca-print-process.pl line 148.
Use of uninitialized value $component in multiplication (*) at superslicer_to_orca-print-process.pl line 148.
Use of uninitialized value in division (/) at superslicer_to_orca-print-process.pl line 148.
Use of uninitialized value $component in multiplication (*) at superslicer_to_orca-print-process.pl line 148.
Translated 'Klipper - Creality Ender-3 S1 Pro (0.4 mm nozzle).ini' to './Klipper - Creality Ender-3 S1 Pro (0.4 mm nozzle).json'.
```
[Klipper - Creality Ender-3 S1 Pro (0.4 mm nozzle).ini.txt](https://github.com/theophile/SuperSlicer_to_Orca_scripts/files/12254789/Klipper.-.Creality.Ender-3.S1.Pro.0.4.mm.nozzle.ini.txt)

MAVProxyUser, 2023-08-03:
Moved from: https://github.com/SoftFever/OrcaSlicer/discussions/1636 to here 

For a sanity check... you may try this repo. 
https://github.com/DIY141/Inland-Filament-Profiles

Make sure on *nix systems that the wildcard is quoted: 
```
Kevins-MacBook-Air:tmp kfinisterre$ perl superslicer_to_orca-filaments.pl --input "Inland-Filament-Profiles/*.ini" --outdir converted --overwrite  
Translated 'Inland-Filament-Profiles/Inland ABS.ini' to 'convertedInland ABS.json'.
Translated 'Inland-Filament-Profiles/Inland CF-NYLON.ini' to 'convertedInland CF-NYLON.json'.
Translated 'Inland-Filament-Profiles/Inland CF-PLA.ini' to 'convertedInland CF-PLA.json'.
Translated 'Inland-Filament-Profiles/Inland Glass PLA.ini' to 'convertedInland Glass PLA.json'.
Translated 'Inland-Filament-Profiles/Inland Lite PLA.ini' to 'convertedInland Lite PLA.json'.
Translated 'Inland-Filament-Profiles/Inland NYLON.ini' to 'convertedInland NYLON.json'.
Translated 'Inland-Filament-Profiles/Inland PETG+.ini' to 'convertedInland PETG+.json'.
Translated 'Inland-Filament-Profiles/Inland PETG.ini' to 'convertedInland PETG.json'.
Translated 'Inland-Filament-Profiles/Inland PLA+.ini' to 'convertedInland PLA+.json'.
Translated 'Inland-Filament-Profiles/Inland PLA.ini' to 'convertedInland PLA.json'.
Translated 'Inland-Filament-Profiles/Inland PP.ini' to 'convertedInland PP.json'.
Translated 'Inland-Filament-Profiles/Inland Pro-PLA.ini' to 'convertedInland Pro-PLA.json'.
Translated 'Inland-Filament-Profiles/Inland Silk PLA.ini' to 'convertedInland Silk PLA.json'.
Translated 'Inland-Filament-Profiles/Inland Tough PLA.ini' to 'convertedInland Tough PLA.json'.
Translated 'Inland-Filament-Profiles/Inland TPU.ini' to 'convertedInland TPU.json'.
Translated 'Inland-Filament-Profiles/Inland Twinkling PLA.ini' to 'convertedInland Twinkling PLA.json'.
Translated 'Inland-Filament-Profiles/Polymaker Polyflex TPU.ini' to 'convertedPolymaker Polyflex TPU.json'.
Translated 'Inland-Filament-Profiles/Polymaker Polylite ASA.ini' to 'convertedPolymaker Polylite ASA.json'.
Translated 'Inland-Filament-Profiles/Polymaker Polylite PC.ini' to 'convertedPolymaker Polylite PC.json'.
Translated 'Inland-Filament-Profiles/Polymaker Polymax PETG.ini' to 'convertedPolymaker Polymax PETG.json'.
Translated 'Inland-Filament-Profiles/Polymaker Polymax PLA.ini' to 'convertedPolymaker Polymax PLA.json'.
Translated 'Inland-Filament-Profiles/Polymaker Polysmooth PVB.ini' to 'convertedPolymaker Polysmooth PVB.json'.
Translated 'Inland-Filament-Profiles/Polymaker Polywood PLA.ini' to 'convertedPolymaker Polywood PLA.json'.
Translated 'Inland-Filament-Profiles/Proto-Pasta HTPLA.ini' to 'convertedProto-Pasta HTPLA.json'.
```

I noticed after attempting to mass convert the Inland profiles, that only 3 of them were accepted by Orca. It seemed to delete all the others! If I manually attempt to import them 0 of them are accepted. However if I place them into ~/Library/Application\ Support/OrcaSlicer/user/default/filament/ they kinda work 

```
Kevins-MacBook-Air:tmp kfinisterre$ ls *json 
Inland ABS.json			Inland Lite PLA.json		Inland PLA+.json		Inland Silk PLA.json		Polymaker Polyflex TPU.json	Polymaker Polymax PLA.json
Inland CF-NYLON.json		Inland NYLON.json		Inland PLA.json			Inland TPU.json			Polymaker Polylite ASA.json	Polymaker Polysmooth PVB.json
Inland CF-PLA.json		Inland PETG+.json		Inland PP.json			Inland Tough PLA.json		Polymaker Polylite PC.json	Polymaker Polywood PLA.json
Inland Glass PLA.json		Inland PETG.json		Inland Pro-PLA.json		Inland Twinkling PLA.json	Polymaker Polymax PETG.json	Proto-Pasta HTPLA.json
Kevins-MacBook-Air:tmp kfinisterre$ cp *json  ~/Library/Application\ Support/OrcaSlicer/user/default/filament/
Kevins-MacBook-Air:tmp kfinisterre$ ls  ~/Library/Application\ Support/OrcaSlicer/user/default/filament/
Inland ABS.json			Inland Lite PLA.json		Inland PLA+.json		Inland Silk PLA.json		Polymaker Polyflex TPU.json	Polymaker Polymax PLA.json
Inland CF-NYLON.json		Inland NYLON.json		Inland PLA.json			Inland TPU.json			Polymaker Polylite ASA.json	Polymaker Polysmooth PVB.json
Inland CF-PLA.json		Inland PETG+.json		Inland PP.json			Inland Tough PLA.json		Polymaker Polylite PC.json	Polymaker Polywood PLA.json
Inland Glass PLA.json		Inland PETG.json		Inland Pro-PLA.json		Inland Twinkling PLA.json	Polymaker Polymax PETG.json	Proto-Pasta HTPLA.json
Kevins-MacBook-Air:tmp kfinisterre$ /Applications/Or
OrcaSlicer.app/         Original Prusa Drivers/ 
Kevins-MacBook-Air:tmp kfinisterre$ /Applications/OrcaSlicer.app/Contents/MacOS/OrcaSlicer 
[2023-08-03 13:32:39.806816] [0x00000001ea739e00] [trace]   Initializing StaticPrintConfigs
```
For some reason only the TPU ones remain after the app is launched: 
```
Kevins-MacBook-Air:tmp kfinisterre$ ls  ~/Library/Application\ Support/OrcaSlicer/user/default/filament/
Inland TPU.json			Polymaker Polyflex TPU.json
```

theophile, 2023-08-03:
> ```
> $ perl superslicer_to_orca-print-process.pl --input
> Klipper - Creality Ender-3 S1 Pro (0.4 mm nozzle).ini  Klipper - Creality Ender-5 S1 (0.4 mm nozzle).ini [...]

Thanks for this. Note though that this particular script only works on the print settings, not printer settings.

MAVProxyUser, 2023-08-03:
whoops! well I bet you know what my next question will be =]

MAVProxyUser, 2023-08-03:
Seems Orca hates it anyway. 
[image]
[Orca+AccControlZero - 0.20 mm NORMAL (0.4 mm nozzle) @CREALITY.ini.txt](https://github.com/theophile/SuperSlicer_to_Orca_scripts/files/12254896/Orca%2BAccControlZero.-.0.20.mm.NORMAL.0.4.mm.nozzle.%40CREALITY.ini.txt)
[Orca+AccControlZero - 0.20 mm NORMAL (0.4 mm nozzle) @CREALITY.json.txt](https://github.com/theophile/SuperSlicer_to_Orca_scripts/files/12254897/Orca%2BAccControlZero.-.0.20.mm.NORMAL.0.4.mm.nozzle.%40CREALITY.json.txt)

MAVProxyUser, 2023-08-03:
I wonder if it is because of the .info files missing? 
```
Kevins-MacBook-Air:default kfinisterre$ find process/
process/
process//0.20mm Standard @Creality Ender5S1 - Copy.json
process//0.20mm Standard @Creality Ender5S1 - Copy.info
```

theophile, 2023-08-03:
Thanks, I'll test with this. FWIW, I never tried Orca's import function (wasn't aware it existed). I just put the converted files directly in the folder. I wonder if the inherits parameter is problematic. None of my profiles have that, and I haven't even been able to figure out how to use inheritance in the first place. But your profile has one so that may be an issue if the parent profile doesn't exist.

theophile, 2023-08-03:
> I wonder if it is because of the .info files missing?
> 
> ```
> Kevins-MacBook-Air:default kfinisterre$ find process/
> process/
> process//0.20mm Standard @Creality Ender5S1 - Copy.json
> process//0.20mm Standard @Creality Ender5S1 - Copy.info
> ```

Shouldn't be. My profiles have converted and loaded fine without them.

theophile, 2023-08-03:
Okay, I tested your [Orca+AccControlZero - 0.20 mm NORMAL (0.4 mm nozzle) @CREALITY.ini.txt](https://github.com/theophile/SuperSlicer_to_Orca_scripts/files/12254896/Orca%2BAccControlZero.-.0.20.mm.NORMAL.0.4.mm.nozzle.%40CREALITY.ini.txt) profile. It converted without issue and I set the output directory to `C:\Users\%USERNAME%\AppData\Roaming\OrcaSlicer\user\default\process\`

The print profile wasn't available when I started OrcaSlicer though. Eventually I realized that the reason was the `compatible_printers` parameter, since my printer isn't named the same thing as what the filter was expecting. I changed that line in the .json to `"compatible_printers" : "",`, and when I restarted OrcaSlicer, the print profile was there and accessible. FWIW, even though the .json file works and OrcaSlicer can use it, it still reports that it cannot _import_ it if I try to use that option. I'll have to look into that.

theophile, 2023-08-04:
Closing this in favor of separate issues broken out from this thread.

closed 2023-08-04 by theophile

theophile, 2023-08-05:
@MAVProxyUser I've refactored the script so that it is now a single script that will autodetect and convert either print profiles or filament profiles. It's in the "unified_script" branch available here: https://github.com/theophile/SuperSlicer_to_Orca_scripts/tree/unified_script

I've also updated the README there accordingly. If you have time to test it out I'd appreciate your feedback/comments. Feel free to open new issues for anything you find, or comment in the existing issues if appropriate.

MAVProxyUser, 2023-08-05:
Cool. I'll try to give it a shot soon!

theophile, 2023-08-07:
> 

I've just added preliminary support for converting printer profiles. It seems to mostly work but there are some edge cases that probably need special handling. One complicating factor is that Prusa/SuperSlicer split some parameters out in a separate "physical printer" config. Ideally I'd be able to pull that in on the fly and parse it but we'll see what's possible.

theophile, 2023-08-17:
@MAVProxyUser , please let me know if you have a chance to test out the script now in the main branch. It can now be run in fully interactive mode with a UI and everything if you run the script with no command-line options. Note that it relies on a few additional perl modules. The README has the cpan line necessary to get all the dependencies installed.

MAVProxyUser, 2023-08-17:
I’ve been super busy brother… it’s on my list for sure!