theophile/SuperSlicer_to_Orca_scripts #2: Filaments script: start_gcode and end_gcode lines don't get translated properly

theophile, 2023-08-03:
Moved from: https://github.com/SoftFever/OrcaSlicer/discussions/1636 to here 

For a sanity check... you may try this repo. 
https://github.com/DIY141/Inland-Filament-Profiles

Make sure on *nix systems that the wildcard is quoted: 
```
Kevins-MacBook-Air:tmp kfinisterre$ perl superslicer_to_orca-filaments.pl --input "Inland-Filament-Profiles/*.ini" --outdir converted --overwrite  
Translated 'Inland-Filament-Profiles/Inland ABS.ini' to 'convertedInland ABS.json'.
Translated 'Inland-Filament-Profiles/Inland CF-NYLON.ini' to 'convertedInland CF-NYLON.json'.
Translated 'Inland-Filament-Profiles/Inland CF-PLA.ini' to 'convertedInland CF-PLA.json'.
Translated 'Inland-Filament-Profiles/Inland Glass PLA.ini' to 'convertedInland Glass PLA.json'.
Translated 'Inland-Filament-Profiles/Inland Lite PLA.ini' to 'convertedInland Lite PLA.json'.
Translated 'Inland-Filament-Profiles/Inland NYLON.ini' to 'convertedInland NYLON.json'.
Translated 'Inland-Filament-Profiles/Inland PETG+.ini' to 'convertedInland PETG+.json'.
Translated 'Inland-Filament-Profiles/Inland PETG.ini' to 'convertedInland PETG.json'.
Translated 'Inland-Filament-Profiles/Inland PLA+.ini' to 'convertedInland PLA+.json'.
Translated 'Inland-Filament-Profiles/Inland PLA.ini' to 'convertedInland PLA.json'.
Translated 'Inland-Filament-Profiles/Inland PP.ini' to 'convertedInland PP.json'.
Translated 'Inland-Filament-Profiles/Inland Pro-PLA.ini' to 'convertedInland Pro-PLA.json'.
Translated 'Inland-Filament-Profiles/Inland Silk PLA.ini' to 'convertedInland Silk PLA.json'.
Translated 'Inland-Filament-Profiles/Inland Tough PLA.ini' to 'convertedInland Tough PLA.json'.
Translated 'Inland-Filament-Profiles/Inland TPU.ini' to 'convertedInland TPU.json'.
Translated 'Inland-Filament-Profiles/Inland Twinkling PLA.ini' to 'convertedInland Twinkling PLA.json'.
Translated 'Inland-Filament-Profiles/Polymaker Polyflex TPU.ini' to 'convertedPolymaker Polyflex TPU.json'.
Translated 'Inland-Filament-Profiles/Polymaker Polylite ASA.ini' to 'convertedPolymaker Polylite ASA.json'.
Translated 'Inland-Filament-Profiles/Polymaker Polylite PC.ini' to 'convertedPolymaker Polylite PC.json'.
Translated 'Inland-Filament-Profiles/Polymaker Polymax PETG.ini' to 'convertedPolymaker Polymax PETG.json'.
Translated 'Inland-Filament-Profiles/Polymaker Polymax PLA.ini' to 'convertedPolymaker Polymax PLA.json'.
Translated 'Inland-Filament-Profiles/Polymaker Polysmooth PVB.ini' to 'convertedPolymaker Polysmooth PVB.json'.
Translated 'Inland-Filament-Profiles/Polymaker Polywood PLA.ini' to 'convertedPolymaker Polywood PLA.json'.
Translated 'Inland-Filament-Profiles/Proto-Pasta HTPLA.ini' to 'convertedProto-Pasta HTPLA.json'.
```

I noticed after attempting to mass convert the Inland profiles, that only 3 of them were accepted by Orca. It seemed to delete all the others! If I manually attempt to import them 0 of them are accepted. However if I place them into ~/Library/Application\ Support/OrcaSlicer/user/default/filament/ they kinda work 

```
Kevins-MacBook-Air:tmp kfinisterre$ ls *json 
Inland ABS.json			Inland Lite PLA.json		Inland PLA+.json		Inland Silk PLA.json		Polymaker Polyflex TPU.json	Polymaker Polymax PLA.json
Inland CF-NYLON.json		Inland NYLON.json		Inland PLA.json			Inland TPU.json			Polymaker Polylite ASA.json	Polymaker Polysmooth PVB.json
Inland CF-PLA.json		Inland PETG+.json		Inland PP.json			Inland Tough PLA.json		Polymaker Polylite PC.json	Polymaker Polywood PLA.json
Inland Glass PLA.json		Inland PETG.json		Inland Pro-PLA.json		Inland Twinkling PLA.json	Polymaker Polymax PETG.json	Proto-Pasta HTPLA.json
Kevins-MacBook-Air:tmp kfinisterre$ cp *json  ~/Library/Application\ Support/OrcaSlicer/user/default/filament/
Kevins-MacBook-Air:tmp kfinisterre$ ls  ~/Library/Application\ Support/OrcaSlicer/user/default/filament/
Inland ABS.json			Inland Lite PLA.json		Inland PLA+.json		Inland Silk PLA.json		Polymaker Polyflex TPU.json	Polymaker Polymax PLA.json
Inland CF-NYLON.json		Inland NYLON.json		Inland PLA.json			Inland TPU.json			Polymaker Polylite ASA.json	Polymaker Polysmooth PVB.json
Inland CF-PLA.json		Inland PETG+.json		Inland PP.json			Inland Tough PLA.json		Polymaker Polylite PC.json	Polymaker Polywood PLA.json
Inland Glass PLA.json		Inland PETG.json		Inland Pro-PLA.json		Inland Twinkling PLA.json	Polymaker Polymax PETG.json	Proto-Pasta HTPLA.json
Kevins-MacBook-Air:tmp kfinisterre$ /Applications/Or
OrcaSlicer.app/         Original Prusa Drivers/ 
Kevins-MacBook-Air:tmp kfinisterre$ /Applications/OrcaSlicer.app/Contents/MacOS/OrcaSlicer 
[2023-08-03 13:32:39.806816] [0x00000001ea739e00] [trace]   Initializing StaticPrintConfigs
```
For some reason only the TPU ones remain after the app is launched: 
```
Kevins-MacBook-Air:tmp kfinisterre$ ls  ~/Library/Application\ Support/OrcaSlicer/user/default/filament/
Inland TPU.json			Polymaker Polyflex TPU.json
```

_Originally posted by @MAVProxyUser in https://github.com/theophile/SuperSlicer_to_Orca_scripts/issues/1#issuecomment-1664609548_

theophile, 2023-08-03:
I confirmed this behavior is because the `start_filament_gcode` line is getting mangled in the conversion and OrcaSlicer chokes on it. Working on sorting that out.

theophile, 2023-08-04:
https://github.com/theophile/SuperSlicer_to_Orca_scripts/commit/1f7c10a6c1b706ba0fdd56efd5ab046d4fbedccf should fix this issue. Config::Tiny was not properly parsing the .ini because the gcode line contained a semicolon between spaces, which is interpreted as a comment so the rest of the line was ignored. I rolled my own ini parser that handles this correctly, with one less dependency being a side benefit.

MAVProxyUser, 2023-08-04:
Seems you broke the directory handling in the fix. =] 
```Kevins-MacBook-Air:filament kfinisterre$ perl /tmp/superslicer_to_orca-filaments.pl --input /tmp/Inland-Filament-Profiles -outdir ./ --overwrite
Use of uninitialized value $superslicer_ini{"slowdown_below_layer_time"} in numeric gt (>) at /tmp/superslicer_to_orca-filaments.pl line 253.
Translated '/tmp/Inland-Filament-Profiles' to './Inland-Filament-Profiles.json'.
```

MAVProxyUser, 2023-08-04:
None the less... Good news. That seems to have worked. 
```
Kevins-MacBook-Air:Inland-Filament-Profiles kfinisterre$ for each in *.ini; do perl /tmp/superslicer_to_orca-filaments.pl --input "$each" -outdir ./ --overwrite; done
Translated 'Inland ABS.ini' to './Inland ABS.json'.
Translated 'Inland CF-NYLON.ini' to './Inland CF-NYLON.json'.
Translated 'Inland CF-PLA.ini' to './Inland CF-PLA.json'.
Translated 'Inland Glass PLA.ini' to './Inland Glass PLA.json'.
Translated 'Inland Lite PLA.ini' to './Inland Lite PLA.json'.
Translated 'Inland NYLON.ini' to './Inland NYLON.json'.
Translated 'Inland PETG+.ini' to './Inland PETG+.json'.
Translated 'Inland PETG.ini' to './Inland PETG.json'.
Translated 'Inland PLA+.ini' to './Inland PLA+.json'.
Translated 'Inland PLA.ini' to './Inland PLA.json'.
Translated 'Inland PP.ini' to './Inland PP.json'.
Translated 'Inland Pro-PLA.ini' to './Inland Pro-PLA.json'.
Translated 'Inland Silk PLA.ini' to './Inland Silk PLA.json'.
Translated 'Inland TPU.ini' to './Inland TPU.json'.
Translated 'Inland Tough PLA.ini' to './Inland Tough PLA.json'.
Translated 'Inland Twinkling PLA.ini' to './Inland Twinkling PLA.json'.
Translated 'Polymaker Polyflex TPU.ini' to './Polymaker Polyflex TPU.json'.
Translated 'Polymaker Polylite ASA.ini' to './Polymaker Polylite ASA.json'.
Translated 'Polymaker Polylite PC.ini' to './Polymaker Polylite PC.json'.
Translated 'Polymaker Polymax PETG.ini' to './Polymaker Polymax PETG.json'.
Translated 'Polymaker Polymax PLA.ini' to './Polymaker Polymax PLA.json'.
Translated 'Polymaker Polysmooth PVB.ini' to './Polymaker Polysmooth PVB.json'.
Translated 'Polymaker Polywood PLA.ini' to './Polymaker Polywood PLA.json'.
Translated 'Proto-Pasta HTPLA.ini' to './Proto-Pasta HTPLA.json'.
Kevins-MacBook-Air:Inland-Filament-Profiles kfinisterre$ cp *.json ~/Library/Application\ Support/OrcaSlicer/user/default/filament/
```
[image]

theophile, 2023-08-04:
Excellent news. I'll fix that error. I know I need better error handling/checks.

MAVProxyUser, 2023-08-04:
May need a hair more massaging. 
[image]

theophile, 2023-08-04:
Yeah, so I think this may end up having to be a known limitation. The script can import the custom g-code from the existing config but it can't rewrite the custom g-code to comply with OrcaSlicer conventions. I mean, in theory it could, but I'm not sure it should. 

It would also take a lot of work. It took me forever to figure out how to translate my filament change g-code to OrcaSlicer conventions. I can't even imagine trying to automate every conceivable possibility.

theophile, 2023-08-04:
I've broken out the existing issues in this thread to their own issues, so closing this one in favor of the others.

closed 2023-08-04 by theophile