theophile/SuperSlicer_to_Orca_scripts #10: Orcaslicer deletes converted Print profiles

Arthur-de-Partuur, 2023-08-22:
I have converted all my Superslicer Print profiles. When finished, I see the JSON files in the ourput folder. When I then start Orcaslicer (V1.6.4) the files get deleted and do not show up in Orcaslicer. I strugled with the filimant and machine files as well but read your remark regarding "compatible printer condition" and solved those more or less.. No such luck for the printer profiles. Any gues to what I can do to make this work?

theophile, 2023-08-22:
In your OrcaSlicer data directory, in the "log" folder, please locate the newest "log.0" file (for the session that deleted the converted "process" files) and attach it to this issue, if you don't mind. That log usually contains clues as to what keys or values OrcaSlicer found offensive.

Also, if you can upload the original .ini file of one of the print profiles that won't work, that would help as well.

Arthur-de-Partuur, 2023-08-22:
[debug_Tue_Aug_22_20_51_55_19452.log.zip](https://github.com/theophile/SuperSlicer_to_Orca_scripts/files/12412357/debug_Tue_Aug_22_20_51_55_19452.log.zip)

Sorry to have to add a couple as I restarted Orcaslicer very often....

Arthur-de-Partuur, 2023-08-22:
Made a new zip with log.0 and input and output file... Something with reason = Invalid value provided for parameter line_width: ?
[debug_Tue_Aug_22_21_38_32_22960.log.zip](https://github.com/theophile/SuperSlicer_to_Orca_scripts/files/12412502/debug_Tue_Aug_22_21_38_32_22960.log.zip)

theophile, 2023-08-22:
All right, I made a few commits that should fix the issue. Please do a git pull, try again, and let me know if that resolved the issue for you.

dareksilver, 2023-08-23:
[superslicer-to-orca-log0-troubleshooting.zip](https://github.com/theophile/SuperSlicer_to_Orca_scripts/files/12414833/superslicer-to-orca-log0-troubleshooting.zip)
I'm running into the same issue as Arthur-de-Partuur, and I'm using your newest commit (downloaded directly from the github).  

I've included the latest log0 file, as well as the ini files of the two printers which keep being deleted when OrcaSlicer opens.  
hopefully this will help sort the issue.

theophile, 2023-08-23:
Thanks, I appreciate the files. I'll take a look and get it ironed out.

Arthur-de-Partuur, 2023-08-23:
[0.30mm SPEED@RatRig (0.6mm nozzle).zip](https://github.com/theophile/SuperSlicer_to_Orca_scripts/files/12419433/0.30mm.SPEED%40RatRig.0.6mm.nozzle.zip
I cleared my perl folder and did a git pull. Ran the script on the 2 attached files. The 0.30mm file converts and is visible in Orcaslicer. The 0.13mm however is converted, NOT deleted upon Orcaslicer start but it is also not visible within Orcaslicer...

theophile, 2023-08-23:
@dareksilver , I pushed a commit that should resolve your issue. Please do a 'git pull' and try again. Let me know if that takes care of it!

theophile, 2023-08-23:
@Arthur-de-Partuur , the issue seems to be that the 0.13mm profile has a `compatible_printers_condition` parameter for which the value is `printer_model=~/.*VC3_.*/ && nozzle_diameter==0.6`. So this process profile will not appear as available unless your selected machine (printer) has "VC3_" in the model and is configured for a 0.6mm nozzle.

theophile, 2023-08-23:
> 

I'm glad to hear you got it working. The thing is, I believe the compatible_printer strings work the same way in OrcaSlicer as they do in SuperSlicer/PrusaSlicer, so _if_ you import all your printers, prints, and filament, the same combinations of profiles _should_ work the same way as they did in SuperSlicer. But I suspect a lot of users are likely to recreate the printer/machine from "scratch" in OrcaSlicer and then only import prints/processes and filament. In that case, a lot of print/process profiles may not show up because they're configured to expect the printer profile that was set up in SuperSlicer.

I'm thinking the best way to handle this is probably to prompt the user and ask them if they want to import the `compatible_printers_condition` value or leave it blank, and explain that they should leave it blank unless they intend to also import the corresponding printer.

Arthur-de-Partuur, 2023-08-23:
Yep, A prompt with a choice seems smart, although you could just strip the `printer_model=~/.*VC3_.*/` altogether. I have no printer configured in SS that contains `VC3_.` at all in its name, so it is probably looking for another place to find this string,...

theophile, 2023-08-23:
> Yep, A prompt with a choice seems smart, although you could just strip the `printer_model=~/.*VC3_.*/` altogether. I have no printer configured in SS that contains `VC3_.` at all in its name, so it is probably looking for another place to find this string,...

I considered the idea of just stripping out the `compatible_printers_condition` key altogether to make sure people's profiles show up as expected, but then for people who import all their profiles and had them linked and organized, stripping out that key would leave their setup messy and disorganized.

dareksilver, 2023-08-23:
> @dareksilver , I pushed a commit that should resolve your issue. Please do a 'git pull' and try again. Let me know if that takes care of it!

Thank you!  I just did an import with the new script, and it successfully transferred the profiles over now. 

Thank you very much.

AndreyShpilevoy, 2023-09-23:
I had the same issue. Fixed by adding 2 values instead of empty into converted json. It were `inner_wall_line_width` and `outer_wall_line_width`.

Vanilla009, 2024-03-13:
@AndreyShpilevoy This worked for me, thank you!

claudermilk, 2024-03-28:
Thank you for that, I was running into the same issue. I also had top_surface_line_width as blank. Filled that with the SS value and it's happy. Probably for any of these just pull the default line width value if nothing else to prevent the files from getting deleted and causing frustration and confusion.

dnlbauer, 2024-10-16:
I had a similar issue and was able to fix it by deleting all variables with an empty string as value from the generated json before opening OrcaSlicer - most were related to `*_line_width`.