theophile/SuperSlicer_to_Orca_scripts #11: Convertion of filament within Linux WSL will only convert one ini File

Sempre, 2023-08-25:
I have tried to convert all my Filaments within an Ubuntu 22.04.3 LTS WSL instance. The converstion was done with access to the SuperSlicer installation on Windows and with the following command:

perl superslicer_to_orca.pl --input /mnt/c/User/<Username>/AppData/Roaming/SuperSlicer/filament/*.ini --outdir ../converted/ --nozzle-size 0.4 --force-output

As Result i got:
[image]

Its is reproducible with the usage of specific Filament names

[image]

An as a proof there is the list of Filaments availabile

[image]

I tried the same directly within windows and with Strawberry Perl which leads to success

[image]

Regards

theophile, 2023-08-25:
Interesting. It looks like the wildcard expansion isn't working right on WSL/Linux. I'll spin up my Ubuntu installation and do some testing. In the meantime, can you let me know what happens if you use the following parameters for the `--input` command?
* `"/mnt/c/User//AppData/Roaming/SuperSlicer/filament/*.ini"` (enclosed in quotation marks)
* `/mnt/c/User//AppData/Roaming/SuperSlicer/filament/` (just the directory path)

theophile, 2023-08-25:
> Tested both and seems to be working. Just for your information, as i started the WSL Linux a few minutes ago there was an update to wsl which i did. It could be a coincidence but i thought i mention it.

Thanks @Sempre. I think I've identified and fixed the problem. Please do a `git pull` and let me know if your original command (with wildcard and no quotes) works as expected now.

Sempre, 2023-08-27:
@theophile 

> Thanks @Sempre. I think I've identified and fixed the problem. Please do a git pull and let me know if your original command (with wildcard and no quotes) works as expected now.

Did a test and it seems to be working now aas expedcted. Thank you for the fast response and fix

theophile, 2023-08-27:
Great! Thanks for the feedback!

closed 2023-08-27 by theophile