theophile/SuperSlicer_to_Orca_scripts #17: Printer profile gets converted to print profile

Cloud-Tickler, 2023-09-28:
Hello together,

I'm trying to convert Prusa XL Single toolhead configuration files to Orcaslicer. 
But even with an exported MK4 printer profile, the script exports a print profile instead of a printer profile. Any hints why this is happening? In the first dialog I select printer and use the attached file.

Thanks a lot

Dennis
[Prusa_MK4_Demo.txt](https://github.com/theophile/SuperSlicer_to_Orca_scripts/files/12754139/Prusa_MK4_Demo.txt)
[image]
[image]

theophile, 2023-09-28:
I've never seen a printer .ini file like that. Do you know where it came from originally? It has over 300 lines, whereas my PrusaSlicer printer profiles only have about 75. It looks like it contains all the parameters for print, printer, and filament profiles. 

The reason the script is treating it like a print profile is that the script looks at the file and counts the parameters according to what profile type they belong to, and then assumes that the file is of whatever type has the most parameter occurrences. Since this input file has all the parameters for all the profile types, it's treating it like a print profile because print profiles have more parameters. 

I can fix this by having the script treat the input file based on whatever directory it found it in, but I'm still curious how this file came to be.

theophile, 2023-09-29:
@Cloud-Tickler , I've pushed a commit that should fix this issue, assuming you use the script in interactive mode like you did originally. FYI, if you were to try to convert this file using the `--input` option, it would still treat like like a print profile because it doesn't seem like there's any way to know what kind of profile this is supposed to be other than the name of the folder it's located in.

Cloud-Tickler, 2023-09-29:
Thanks a lot, great work. I tried the new version and was able to generate a printer config for the MK4 file that succesfully showed up in OrcaSlicer.
I`m using PrusaSlicer 2.6.1 to export the configuration file. Is there a different way? I added a video.
I also managed to export a config file for the Prusa XL single toolhead that shows up in OrcaSlicer. I will do some testing with this. Maybe you wann add the Prusa XL in the selection dialog for the interactive mode as well.

https://github.com/theophile/SuperSlicer_to_Orca_scripts/assets/127838436/3231fb70-ffa0-4739-ae95-c52a9422baa1
[image]
[Prusa XL Single 060.txt](https://github.com/theophile/SuperSlicer_to_Orca_scripts/files/12762254/Prusa.XL.Single.060.txt)
[Prusa_MK4_Demo.txt](https://github.com/theophile/SuperSlicer_to_Orca_scripts/files/12762255/Prusa_MK4_Demo.txt)

theophile, 2023-09-29:
Interesting. It shouldn't be necessary to export the config like that at all. You should just be able to run the script and it should find those source .ini files in that directory.

Cloud-Tickler, 2023-09-29:
As far as I can see, initially this folder is empty if you don't run an export. All system printers that you configure within PrusaSlicer are stored at a different location in one big .ini file.

C:\Users\dh\AppData\Roaming\PrusaSlicer\vendor

If you've run an export, printers will show up as user presets.

[image]
[image]

[PrusaResearch.txt](https://github.com/theophile/SuperSlicer_to_Orca_scripts/files/12771840/PrusaResearch.txt)

Cloud-Tickler, 2023-09-29:
Sadly I got another issue. I don't know what I've changed, but the skript keeps stopping and I can't get to the new dialogue where you choose the printer .
[image]

https://github.com/theophile/SuperSlicer_to_Orca_scripts/assets/127838436/54906449-43a7-4b0d-b34c-e6d18c24a659

theophile, 2023-09-29:
> As far as I can see, initially this folder is empty if you don't run an export. All system printers that you configure within PrusaSlicer are stored at a different location in one big .ini file.

Ah, I see. You're converting the built-in system presets. I have to admit that this use case didn't even occur to me, although it's certainly not unreasonable. I assumed that OrcaSlicer would also have these built-in presets so no need to convert, though I have not actually checked.

theophile, 2023-09-29:
> Sadly I got another issue. I don't know what I've changed, but the skript keeps stopping and I can't get to the new dialogue where you choose the printer .

This is happening because your OrcaSlicer installations does not have any physical printers set up. I definitely need to implement proper error handling for this situation, but as noted in the [README](https://github.com/theophile/SuperSlicer_to_Orca_scripts#a-note-about-printer-profiles), you need to set up and configure a printer from one of OrcaSlicer's system presets _before_ you run this script, or else the converted printer profile will not have the ability to use or configure network access/gcode uploading.

Cloud-Tickler, 2023-09-29:
> Ah, I see. You're converting the built-in system presets. I have to admit that this use case didn't even occur to me, although it's certainly not unreasonable. I assumed that OrcaSlicer would also [...]

Yes, exactly. The printer hasn't been imported into OrcaSlicer yet, but I want to use it.  I guess this is not the only printer applicable. So, thanks again for modifying the skript.

Cloud-Tickler, 2023-09-29:
> This is happening because your OrcaSlicer installations does not have any physical printers set up. I definitely need to implement proper error handling for this situation, but as noted in the [...]

Sorry, seems I have been cleaning up the folders a bit too much because I lost track. Works perfectly again. Thanky you so much.

closed 2023-10-17 by theophile