theophile/SuperSlicer_to_Orca_scripts #19: Ironing and Elephant foot

B0IIE, 2023-10-15:
I used your script to convert my settings from PS 2.6.1 to OS 1.7.0 and noticed that after importing in all my profiles Ironing was turned on even though it was disabled in PS and that Elephant foot compensatation was set to 0 even though I had it set to 0,2 mm in PS.

theophile, 2023-10-15:
Thanks for this. Can you post a source profile where you experienced this issue?

B0IIE, 2023-10-15:
Here´s an example.
[0.20mm NORMAL Ender3_3_3_3.zip](https://github.com/theophile/SuperSlicer_to_Orca_scripts/files/12910071/0.20mm.NORMAL.Ender3_3_3_3.zip)

Creo2005, 2023-10-16:
> Thanks for this. Can you post a source profile where you experienced this issue?

When I imported from SS, it was the same. Ironing was turned on. In ss in off.

theophile, 2023-10-17:
I've pushed a commit that should address the ironing problem. Let me know if not. I'll take a look at the elephant foot compensation next.

theophile, 2023-10-17:
Elephant foot compensation should be fixed now too. I had failed to notice that PrusaSlicer and SuperSlicer have different names for the same parameter.

closed 2023-10-18 by B0IIE