theophile/SuperSlicer_to_Orca_scripts #21: Use of uninitialized value

vin750, 2023-12-20:
Hi there,

I am trying to covert a custom profile that uses as post processor for a Flashforge Creator Pro 2 but I am getting some errors. This basically lets me use a dual extruder printer by putting the gcode through a post processor and each extruder is basically seen as a different printer inside the slicer.

Would you be able to assist me with this issue?

`Use of uninitialized value $header_line in concatenation (.) or string at superslicer_to_orca.pl line 258.
Use of uninitialized value $header_line in concatenation (.) or string at superslicer_to_orca.pl line 258.
Use of uninitialized value $header_line in concatenation (.) or string at superslicer_to_orca.pl line 258.
Use of uninitialized value $header_line in concatenation (.) or string at superslicer_to_orca.pl line 258.
Use of uninitialized value $header_line in concatenation (.) or string at superslicer_to_orca.pl line 258.
Use of uninitialized value $status{"ini_type"} in string eq at superslicer_to_orca.pl line 1386.
Use of uninitialized value $status{"ini_type"} in ucfirst at superslicer_to_orca.pl line 1396.
Use of uninitialized value $status{"ini_type"} in string eq at superslicer_to_orca.pl line 1386.
Use of uninitialized value $status{"ini_type"} in ucfirst at superslicer_to_orca.pl line 1396.
Use of uninitialized value $status{"ini_type"} in string eq at superslicer_to_orca.pl line 1386.
Use of uninitialized value $status{"ini_type"} in ucfirst at superslicer_to_orca.pl line 1396.
Use of uninitialized value $status{"ini_type"} in string eq at superslicer_to_orca.pl line 1386.
Use of uninitialized value $status{"ini_type"} in ucfirst at superslicer_to_orca.pl line 1396.
Use of uninitialized value $status{"ini_type"} in string eq at superslicer_to_orca.pl line 1386.
Use of uninitialized value $status{"ini_type"} in ucfirst at superslicer_to_orca.pl line 1396.
                                   CONVERSION SUMMARY

 Files Converted
.--------------+-------------------------------------+------------+--------------------.
| Source File  |  Profile Name                       | Converted? | Error              |
| Generated By |                                     |            |                    |
+--------------+-------------------------------------+------------+--------------------+
| Unknown      | CreatorPro2                         | NO         | Unsupported slicer |
| Unknown      | CreatorPro2 Dual extrusion          | NO         | Unsupported slicer |
| Unknown      | CreatorPro2 E1 right only extrusion | NO         | Unsupported slicer |
| Unknown      | CreatorPro2 E2 left only extrusion  | NO         | Unsupported slicer |
| Unknown      | CreatorPro2 E1 right E2 Ditto       | NO         | Unsupported slicer |
'--------------+-------------------------------------+------------+--------------------'`

`>perl -v

This is perl 5, version 38, subversion 0 (v5.38.0) built for MSWin32-x64-multi-thread
`

Here is the link to the Prusaslicer profile I am trying to convert.

[https://github.com/Jacotheron/FlashForge-CreatorPro2-PS-Profile/blob/main/vendor/FlashForge.ini](url)

Thanks in advance

theophile, 2023-12-31:
I played with your config bundle and pushed some commits that should help the script process it, but there are two caveats (so far).

First, the script expects the first line of the file to be in the format:

`# generated by PrusaSlicer 2.7.1 on 2023-12-30 at 19:47:00 UTC`

It **_must_** start with a hash, then a space, then the phrase "generated by", then another space, then the name of the slicer. The script needs to be able to identify the source slicer so it knows what conventions to follow when trying to translate the settings. So you will need to modift the first line of your config bundle accordingly.

Second, OrcaSlicer doesn't appear to permit profile names to contain illegal characters such as the asterisks you use frequently. The script will try to convert these and will try to preserve the names, but I don't think OrcaSlicer will respect them. This means that the inheritance structure you've set up probably will not work.

theophile, 2024-01-04:
I think you would use whichever matches your printer's firmware.

closed 2024-01-04 by theophile