theophile/SuperSlicer_to_Orca_scripts #25: How TF do you use this?

TheVoronModder, 2024-02-09:
WTF is Perl?

And how does someone who does literally NO coding make this work?

Thank you.

Please have patience with me.

theophile, 2024-02-09:
There are installation and usage instructions here: https://github.com/theophile/SuperSlicer_to_Orca_scripts#installation

Perl is an interpreted programming language, which means that a program's "source code" doesn't have to be compiled into an executable binary. Instead, it can be executed directly by feeding it into a Perl interpreter. The first thing you need to do is install a Perl interpreter. If you're using Windows, the installation instructions contain a link where you can download a Windows Perl interpreter.

Once that's installed, just follow the rest of the instructions. In a nutshell, you will need to use the command line to call your Perl interpreter and pass this script to it as an argument.

closed 2024-02-09 by theophile

TheVoronModder, 2024-02-10:
So I’m not understanding how to get stawberry Perl opened at all. I downloaded and installed the software however I don’t know what or where a Perl console is? Sent from my iPhone

theophile, 2024-02-10:
Click the Windows icon/start menu. In the search box, type `cmd` and hit enter. That's the command prompt. If you don't understand how it works or how to navigate your system with it, you may have to Google it. In a nutshell, you'll need to change the working directory (`cd`) to wherever you downloaded this script, and then just follow the usage instructions in the repo.

TheVoronModder, 2024-02-11:
Okay that makes sense. I have 6 hdd so I’ll have to change the directory to the correct drive. Thank you so much for your patience and understanding! Sent from my iPhone

TheVoronModder, 2024-02-11:
I am having errors with #3

**Clone this repository or download the script directly from GitHub.

git clone https://github.com/theophile/SuperSlicer_to_Orca_scripts.git**

theophile, 2024-02-11:
You can just download the script directly: https://raw.githubusercontent.com/theophile/SuperSlicer_to_Orca_scripts/main/superslicer_to_orca.pl

TheVoronModder, 2024-02-11:
I should have explained more, I am stuck with this part:

[image]

TheVoronModder, 2024-02-11:
[image]

I figured it out! Thank you!