theophile/SuperSlicer_to_Orca_scripts #26: Can't Located Object method "new" jason line 1734

TheVoronModder, 2024-02-11:
[image]

theophile, 2024-02-11:
Thanks. That was my fault. It should be fixed now.

closed 2024-02-11 by theophile

TheVoronModder, 2024-02-12:
Haha no big deal. Not bad for someone who does not have any coding experience other than Java and html lol. Learning about Perl this week. It’s very interesting. Thank you! Sent from my iPhone

TheVoronModder, 2024-02-12:
[image]

Still getting the issue, I will re-download the script and re-try

TheVoronModder, 2024-02-12:
After a successful conversion, opening orca slicer, closing it, locating converted file then manually importing into Orca the preset is not there...

[image]

Any ideas?

theophile, 2024-02-12:
This is probably because of the `compatible_printers` condition in the original profile. If your source profile contained a `compatible_printers` condition and the printer (machine) in OrcaSlicer doesn't have exactly the same name as the value for that field, then the prints (processes) won't show up because OrcaSlicer thinks they're not compatible. Try running the script again but select only one print profile to convert as a test. During the conversion process you'll be alerted if it detects a `compatible_printers` condition and you will be prompted to either keep or discard it. Choose to discard it and then continue the conversion process. After it finished, fire up OrcaSlicer again and see if that print (process) profile appears. If so, then you can either do the same with all your other profiles, or if you have multiple printers and want to maintain the `compatible_printers` condition, you may need to manually edit the resulting .json files to make sure the field contains the exact name of the associated printer.