theophile/SuperSlicer_to_Orca_scripts #27: This tool doesn't work

andree0, 2024-02-27:
Command don't return any output and don't create JSON file.

Command: `perl superslicer_to_orca.pl --input "~/Downloads/24.02.27\ PrusaSlicer_config_bundle.ini" --outdir ./ --force-output --on-existing overwrite`

I tried to run the script with difference parameters and result always is the same, no file and no output.

Perl: v5.34.0

theophile, 2024-02-27:
Does this work?

`perl superslicer_to_orca.pl --input "~/Downloads/24.02.27 PrusaSlicer_config_bundle.ini" --outdir ./ --force-output --on-existing overwrite`

andree0, 2024-02-29:
> Does this work?
> 
> `perl superslicer_to_orca.pl --input "~/Downloads/24.02.27 PrusaSlicer_config_bundle.ini" --outdir ./ --force-output --on-existing overwrite`

Changed file name is not solution. :smile: 
I have already tried with file name without whitespace.
I think It's the worst what can be for programmer, no output and no message :slightly_smiling_face: 

I only wanted to help my friend and converted one file. I don't know perl so I am not going to fix it but I opened this issue because I hope that You will improve this code and It will work in the future.

theophile, 2024-02-29:
Can you please post a copy of the config bundle you are trying to convert so I can test?

theophile, 2024-03-01:
I was able to reproduce the behavior you described by specifying an input file that either does not exist or is not a regular file. I have pushed a commit that should output useful information when this happens. Please either do a `git pull` or download a new copy of the script and give it another try to confirm this is the issue.

andree0, 2024-03-01:
[24.02.27_PrusaSlicer_config_bundle.txt](https://github.com/theophile/SuperSlicer_to_Orca_scripts/files/14457912/24.02.27_PrusaSlicer_config_bundle.txt)

I did it how You wrote. It still doesn't work. I'm attaching file which I want to convert but for this message I changed file extension from `.ini` to `.txt`. Github's interface doesn't allow me to attach file with `.ini` ext.
I also run command for `.txt` extension and I already get message `24.02.27_PrusaSlicer_config_bundle.txt is not a .ini file!`. It's good because I know that something is wrong but with `.ini` extension I still don't get any message.

theophile, 2024-03-01:
I was able to convert the profiles successfully with the command:
`perl Documents\SuperSlicer_to_Orca_scripts\superslicer_to_orca.pl --input "Downloads\24.02.27 PrusaSlicer_config_bundle.ini"  --outdir ./ --force-output --on-existing overwrite`

Here's the conversion summary:

```
                           CONVERSION SUMMARY

Print Files Converted
.--------------+------------------------------------------+------------.
| Source File  | Print Profile Name                       | Converted? |
| Generated By |                                          |            |
+--------------+------------------------------------------+------------+
| PrusaSlicer  | 0.15mm STRUCTURAL @MK4IS 0.6 - OdÅ,Ä.cz- | YES        |
|              | ono                                      |            |
| PrusaSlicer  | 0.20mm NORMAL @Artillery - Kopia         | YES        |
| PrusaSlicer  | 0.20mm SPEED @MK4IS 0.6 - OdÅ,Ä.czono    | YES        |
| PrusaSlicer  | 0.25mm STRUCTURAL @MK4IS 0.6 - OdÅ,Ä.cz- | YES        |
|              | ono                                      |            |
| PrusaSlicer  | 0.32mm SPEED @MK4IS 0.6 - CARGO          | YES        |
| PrusaSlicer  | 0.32mm SPEED @MK4IS 0.6 - OdÅ,Ä.czono    | YES        |
| PrusaSlicer  | 0.32mm SPEED @MK4IS 0.6 - julia nessa m- | YES        |
|              | akiety                                   |            |
| PrusaSlicer  | 0.32mm lampka hex                        | YES        |
| PrusaSlicer  | 0.32mm vase mode @MK4IS 0.6 - OdÅ,Ä.czo- | YES        |
|              | no                                       |            |
| PrusaSlicer  | 0.4mm SPEED 0.6                          | YES        |
| PrusaSlicer  | Ender 0.4 default                        | YES        |
| PrusaSlicer  | Ender 0.4 plozy tpu                      | YES        |
| PrusaSlicer  | Ender 0.6 default                        | YES        |
| PrusaSlicer  | Ender 0.6 pakiet domina                  | YES        |
| PrusaSlicer  | Inventure 0.4 RAFT                       | YES        |
| PrusaSlicer  | Inventure 0.4 default                    | YES        |
| PrusaSlicer  | SWx2 0.6 oslonka doniczki 0.28mm         | YES        |
| PrusaSlicer  | SWx2 0.6 oslonka doniczki 0.32mm         | YES        |
'--------------+------------------------------------------+------------'

Printer Files Converted
.--------------+----------------------------------+------------.
| Source File  | Printer Profile Name             | Converted? |
| Generated By |                                  |            |
+--------------+----------------------------------+------------+
| PrusaSlicer  | Ender 0.4                        | YES        |
| PrusaSlicer  | Ender 0.6                        | YES        |
| PrusaSlicer  | Original Prusa i3 MK3 - Kopiuj   | YES        |
| PrusaSlicer  | SWx2 stock 0.6mm klipper         | YES        |
| PrusaSlicer  | SWx2 stock 0.6mm test kalibracji | YES        |
| PrusaSlicer  | Zortrax Inventure                | YES        |
'--------------+----------------------------------+------------'

Filament Files Converted
.--------------+------------------------------+------------.
| Source File  | Filament Profile Name        | Converted? |
| Generated By |                              |            |
+--------------+------------------------------+------------+
| PrusaSlicer  | HMF ABS                      | YES        |
| PrusaSlicer  | HMF PCABS                    | YES        |
| PrusaSlicer  | HMF PLA                      | YES        |
| PrusaSlicer  | PETG Ender                   | YES        |
| PrusaSlicer  | PETG Inventure               | YES        |
| PrusaSlicer  | PLA Ender                    | YES        |
| PrusaSlicer  | PRO3D ABS                    | YES        |
| PrusaSlicer  | Rosa3D Biowood               | YES        |
| PrusaSlicer  | Rosa3D PCTG                  | YES        |
| PrusaSlicer  | Rosa3D PETG                  | YES        |
| PrusaSlicer  | Rosa3D PLA Magic Silk        | YES        |
| PrusaSlicer  | Rosa3D PLA Starter           | YES        |
| PrusaSlicer  | Rosa3D PLA starter Satin red | YES        |
| PrusaSlicer  | Spectrum wood natural        | YES        |
| PrusaSlicer  | TPU Ender                    | YES        |
| PrusaSlicer  | eryone PLA matt              | YES        |
'--------------+------------------------------+------------'
```

I'm not sure why it's not working on your system. I will try to test it on Linux and see if that makes a difference.

andree0, 2024-03-01:
I am working on Pop!_OS which based on Ubuntu. Maybe It 's the reason why this code doesn't work.

andree0, 2024-03-01:
If You can, please send the JSON file to the e-mail address `kontakt@drzewniane.pl`.

btw, I removed the message with the file because I don't want to share it for everyone.
Thanks for your help and your code, keep going. :wink:

jhyland87, 2024-03-03:
You really should have written this in Python or something like that. I didn't realize people were still using it for new scripts. Just installing the dependency perl modules is taking absolutely forever and making a mess. And many of the modules this requires are modules that are built into python.
Im trying to execute it now though once everything's installed. Thank you for creating this.

Gozania, 2024-03-24:
This did not work for me either.  Followed the steps provided in installation.  I run perl superslicer_to_orca.pl after installation and installation of windows git 

and am met with no directory.  All of my installations are in default locations.
[image]

I am not skilled enough to find out why it doesnt work.

theophile, 2024-03-24:
1. `git clone https://github.com/theophile/SuperSlicer_to_Orca_scripts.git`
2. `cd SuperSlicer_to_Orca_scripts`
3. `perl superslicer_to_orca.pl`

Gozania, 2024-03-25:
Will try this later and will respond.  Ty for the extra instructions.

Gozania, 2024-03-25:
That worked!  TY!

closed 2024-03-26 by theophile