theophile/SuperSlicer_to_Orca_scripts #31: Can't generate file and silent log.

cropse, 2024-05-06:
Download file from https://store.anycubic.com/pages/firmware-software
File: https://drive.google.com/file/d/1D-iqxPlP9TRq5NXVWtQRBSRd3YL2Ip79/view

Clone the repo and create output dir in repo, also rename file to remove all space to `_`, runing pure docker.
```
docker run -it --rm -v '.\:/ok' perl:5.36 sh

apt-get update
apt-get install libcam-pdf-perl

cpan Getopt::Long File::Basename File::Glob File::HomeDir Path::Class Path::Tiny String::Escape Term::Choose Term::Form::ReadLine Text::SimpleTable JSON::XS
cd /ok/SuperSlicer_to_Orca_scripts

perl superslicer_to_orca.pl --input ./Anycubic_Kobra_2_Plus_configuration_setting.ini --outdir ./output --on-existing skip
```
It end up nothing happen, no output log and no file generated.

White3rabbit, 2024-05-07:
oh no.. realy? i want to do the same

White3rabbit, 2024-05-07:
my next printer is a bambu lab x1 carbon combo or something. The possibilities for this printer are limited in terms of profiles and configurations, as they are too copied to be integrated. The "other" printers have much more support!

respectTheCode, 2024-07-09:
This works but you need to use the `--force-output` flag.