theophile/SuperSlicer_to_Orca_scripts #36: Port to Perl 5.40

reneas, 2025-01-26:
When trying to run the script on my system i get this error message:

```
Can't load '/PATH/TO/PERL/perl5/lib/perl5/x86_64-linux-thread-multi/auto/JSON/XS/XS.so' for module JSON::XS: libperl.so.5.38: Kann die Shared-Object-Datei nicht öffnen: Datei oder Verzeichnis nicht gefunden at /usr/share/perl5/XSLoader.pm line 94.
 at /PATH/TO/PERL/perl5/lib/perl5/x86_64-linux-thread-multi/JSON/XS.pm line 1821.
Compilation failed in require at ./superslicer_to_orca.pl line 16.
BEGIN failed--compilation aborted at ./superslicer_to_orca.pl line 16.
```

After a bit of research i think this is caused by anewer perl version that is present on my system, i only have perl 5.40 installed.
maybe you can port this script?

theophile, 2025-01-29:
It looks like your perl installation may be messed up, you have conflicting versions installed, or maybe don't have dependencies installed. Try:
`cpan Getopt::Long File::Basename File::Glob File::HomeDir Path::Class Path::Tiny String::Escape Term::Choose Term::Form::ReadLine Text::SimpleTable JSON::XS`

reneas, 2025-01-31:
I get this:

```
CPAN: Module::CoreList loaded ok (v5.20250120)
Getopt::Long is up to date (2.58).
File::Basename is up to date (2.86).
File::Glob is up to date (1.42).
File::HomeDir is up to date (1.006).
Path::Class is up to date (0.37).
Path::Tiny is up to date (0.146).
String::Escape is up to date (2010.002).
Term::Choose is up to date (1.767).
Term::Form::ReadLine is up to date (0.558).
Text::SimpleTable is up to date (2.07).
JSON::XS is up to date (4.03).
```
The module JSON::XS is present and up to date
It seems like somehow `libperl5.38` is used but i dont have this installed. Should i downgrade? Any other idea?

theophile, 2025-01-31:
Best I can figure is that you have conflicting versions of perl installed. You said you have 5.40 installed but the interpreter you're executing appears to be looking for the shared library for 5.38. Maybe try `perl -v` to see what version you're actually executing. And then maybe see if you have other version of the perl binary on your system, and if so, call it with its full path when executing the script.

reneas, 2025-01-31:
`perl -v` gives 5.40 thats why I'm saying i have 5.40 installed. And i can't find anything with `locate perl | fzf` looking for `5.38`. Just finding `/var/lib/snapd/snap/gnome-42-2204/176/usr/lib/x86_64-linux-gnu/perl/5.34.0` but I guess that has nothing to do with this. after reinstalling perl with `sudo dnf reinstall perl` the error still occurs :(

I can't find any trace of `libperl5.38` on my system.

But in the meantime I found the solution! It's [here](https://stackoverflow.com/a/29468484/23195983)

So what I did:

```bash
cpan App::cpanminus
cpanm --uninstall JSON::XS
cpan JSON::XS
```

And now it works again!

closed 2025-01-31 by reneas