theophile/SuperSlicer_to_Orca_scripts #37: Unable to run script. Compilation errors.

wonderbread24, 2025-01-28:
Attempting to run the script by following the instructions. I've tried on both Strawberry Perl 5.40.0.1 and 5.38.2.2 with the same result. 

`E:\3D Printer>perl superslicer_to_orca.pl
Bareword found where operator expected (Missing semicolon on previous line?) at superslicer_to_orca.pl line 11, near "data"
Can't modify numeric lt (<) in scalar assignment at superslicer_to_orca.pl line 11, near "data"
syntax error at superslicer_to_orca.pl line 11, near "data"
Execution of superslicer_to_orca.pl aborted due to compilation errors.`

I'm not sure what this error means. I'm using Windows 11.

theophile, 2025-01-29:
Dup of #12

closed 2025-01-29 by theophile

wonderbread24, 2025-01-29:
Can you help me understand how it’s a dup?

I don’t see a solution? I have downloaded the script directly from GitHub, twice with the same error…

theophile, 2025-01-29:
Open the script in notepad and you should see that it's not actually a perl script.

wonderbread24, 2025-01-29:
Ok…so help me out here I’m not a programmer of any sorts. If it’s not a script why is it called a script? How is it supposed to work if there’s no script in the “script”

What’s the issue if the script that’s not a script won’t run?

theophile, 2025-01-29:
The issue is that you did not download the actual perl script. If you're going to try to download just the script file without cloning the repo, you need to download the "raw" version as noted in my response in #12.