theophile/SuperSlicer_to_Orca_scripts #39: Filename format in orca has wrong syntax when you change nozzle size.

aymericblok, 2025-01-30:
When i changed nozzle size (maybe during your script already, which is awesome)
The "filename-format" wat:

`{input_filename_base}_{nozzle_diameter{0}}n_{layer_height}mm_{filament_type{0}}_{printer_model}_{print_time}.gcode`

but it should have [ ] instead of { } at the array identifiers.

`{input_filename_base}_{nozzle_diameter[0]}n_{layer_height}mm_{filament_type[0]}_{printer_model}_{print_time}.gcode`

Without changing this manually, slicing went great, but i couldn't export the .gcode
Giving error message: `Processing of the filename_format template failed.
Parsing error at line 1: Referencing a vector variable when scalar is expected
{input_filename_base}_{nozzle_diameter{0}}n_{layer_height}mm_{filament_type{0}}_{printer_model}_{print_t
ime}.gcode`

Non programmers would have a hard time fixing this!
Thanks! and sorry if this isn't due to your program, which is awesome that you made it! damn

theophile, 2025-01-31:
This looks like it relates to the formatting of custom gcode in the converted file. Does the original (source) file have square brackets that the script is changing into curly braces during conversion? Or are you saying that the script should detect and convert the curly braces in the original file to square brackets because that's what OrcaSlicer needs?

aymericblok, 2025-02-01:
it looks like the original has the [] when opening in prusa slicer, and when converting to orca with your script, it ends up being {}.

Here is the original file: "https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0-alpha1" it's the 16 minute bonkers benchy gcode. And in prusa slicer choose "import -> import config"

Then in prusa the string is: "{input_filename_base}_{nozzle_diameter[0]}n_{layer_height}mm_{filament_type[0]}_{printer_model}_{print_time}.gcode"

and after your convert it's: "{input_filename_base}_{nozzle_diameter{0}}n_{layer_height}mm_{filament_type{0}}_{printer_model}_{print_time}.gcode"

That's why i ask you :)

theophile, 2025-02-01:
I'm not clear on what you're reporting. The file you referred to is a gcode file. I'm not sure what that has to do with this script.

apparle, 2025-04-22:
FWIW I've fixed this as part of my PR https://github.com/theophile/SuperSlicer_to_Orca_scripts/pull/43

ipatch, 2025-07-16:
can confirm, i encountered this error message when attempting to export / print a file after a successful slice in orca slicer, after importing my filament, print, and printer settings from prusa slicer.